bigtreetech/BTT002-V1.0 #2: missing schematics

mstewart1987, 2020-03-17:
missing the cover plate for the hotbed cables.  only supplied cover plate for extruder.

thisiskeithb, 2020-06-09:
I sent a note to Biqu/BigTreeTech about this missing heatbed cable clip since Prusa's doesn't fit in that spot.

In the meantime, Juan Puchol made a pretty awesome [Universal BTT002 Box](https://www.thingiverse.com/thing:4231370). Be sure to read the readme for more info, including links to the other parts of the box.

Your title mentions missing schematics and [those have been uploaded](https://github.com/bigtreetech/BTT002-V1.0/blob/7c1048484338716e2f823528319dfaf37559c367/Hardware/BTT002%20V1.0SCH.pdf).

3d-gussner, 2020-06-11:
@thisiskeithb Yes and no, the schematics are old and don't have the new STM32 chip with 1024
Also i cannot find the connection to the diag and index pin in there.