bigtreetech/BTT002-V1.0 #3: firmware flashing over sd with TFT35 V3.0 does not work

Lino77, 2020-04-01:
firmware flashing over sd with TFT35 V3.0 does not work. how can you get the firmware on the board. without changing the display every time

michaelshulichenko, 2020-05-15:
in my case flashing works only with RepRap Discount Smart Controller (2004)
it does not work with Full Graphic Smart Controller (12864) and BTT TFT 35 v2.0
i am flashing it with stlink (be careful to not erase bootloader, marlin code starts from 0x08008000 )

thisiskeithb, 2020-05-16:
The major pitfall is that once you update the firmware to use a different LCD than a stock MK3S 2004 character style (`REPRAP_DISCOUNT_SMART_CONTROLLER`), you're stuck using an ST-Link for future updates.

chzj333, 2020-05-18:
> in my case flashing works only with RepRap Discount Smart Controller (2004)
> it does not work with Full Graphic Smart Controller (12864) and BTT TFT 35 v2.0
> i am flashing it with stlink (be careful to not erase bootloader, marlin code starts from 0x08008000 )

I removed the motherboard from the machine, only used USB power, and connected 12864, then it was updated, maybe you can try it.

thisiskeithb, 2020-06-09:
I spoke with an engineer at Biqu/BigTreeTech, they provided an answer as to why flashing from a TFT doesn't work:

> Our engineers said the LCD mode in TFT35/E3 V3.0 is a LCD simulator, the EXP1 and EXP2 cable are connected to MCU on the display, not connected to SD slot. So the BTT002 cannot flash firmware by SD slot on TFT35 display.

You'll have to use a traditional full graphic or text LCD to update the firmware on a BTT002.

I suggested they add an onboard SD card reader, similar to their other boards, so users won't run into this issues when using TFTs.

iShavgula, 2020-08-14:
Hello,

I'm totally new to this and I've got:
- BTT002
- TFT35 v3.0

Nothing else: other displays, ST-link, etc..

The display turns on, shows me two choices to proceed: `Marlin mode` and `Touch mode` but I can not click on any of them. 

I'm having trouble updating the board firmware as mentioned in this issue. Are there any updates on this ? If I have to buy something additional(ST-Link for example) what would be the best choice ?

Thanks

iShavgula, 2020-08-15:
Hello again,

I got the ST-Link and I'm using OS X. Could you please tell me to which address I should flash the firmware?
Thanks

iShavgula, 2020-08-15:
Hello,

I got the ST-Link, flashed the new firmware but I still got the same results.
Here the modifications I applied to the BTT002 Marlin configuration:
- uncomment `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`
- set `LCD_INFO_SCREEN_STYLE` to 0 instead of 1

iShavgula, 2020-08-15:
I connected only the serial port and the now the display works, but only in touch mode.

closed 2021-01-04 by Lino77