bigtreetech/BTT002-V1.0 #5: Flashing BTT002-V1.0 doesn't write CUR

ogland, 2020-05-18:
As the title reads, it seems to flash fine but the BIN file persists. This is using different 128 mb cards from BTT and both with 2004 LCD and 12864 LCD.

Same behaviour if powered through USB or externally.

Also as a sidenote, one of the motors doesn't move properly. I'm not sure if that's related to some kind of incomplete write of firmware or if it's a physical error. I moved drivers and motors around and it's linked to the specific port/pins.

thisiskeithb, 2020-06-09:
> As the title reads, it seems to flash fine but the BIN file persists. This is using different 128 mb cards from BTT and both with 2004 LCD and 12864 LCD.

Does a .cur file exist on the SD card? I found that I had to delete the .cur file after flashing since the bootloader doesn't overwrite it. I also ran into this issue on some other BigTreeTech boards.

Afterschnitzel, 2020-12-22:
I can confirm it. FW will flash, but firmware.bin is not renamed as FIRMWARE.CUR

ebraiman, 2021-01-01:
@Afterschnitzel May I ask the process of loading the firmware on the BTT002?  Are you loading via a SD card through a LCD or TFT?

Afterschnitzel, 2021-01-02:
I do'nt use TFT, because I am running my printer on Klipper. I made flashing proccess via SD card through Fysetc 12864 Mini Neopixel.