bigtreetech/BTT002-V1.0 #7: TFT35 v3.0 No printer attached!

iShavgula, 2020-08-15:
Hello,

I'm trying to connect BTT TFT35 v3.0 to the BTT002 board and I'm getting this error: `No printer attached`.

Here are the modifications I applied to the original BTT002 Marlin setup:
- uncomment REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
- set LCD_INFO_SCREEN_STYLE to 0 instead of 1

I tried to play with these settings, but in vain:
- BAUDRATE 115200
- SERIAL_PORT 3
- SERIAL_PORT_2 -1

What am I missing?

Thanks

iShavgula, 2020-08-16:
I plugged in only the TFT cable without reset pin and the printer got detected. 

I'd not say the issue is resolved as I still can't use the marlin mode.

DoNotGood, 2020-10-06:
Hello iShavgula,
today I got my tft35 (I also have Prusa MK3s with BTT002 V1.0) and I have exactly the same problems with display!

Do you have found a solution, which works well in Marlin- and in TFT-mode?

I could not choose Marlin-mode.

I connected EXP1+2 and TFT, do I also have to connect +5V and GND (Power Supply)?

Both Boards are made by BigTreeTech, so it must have been tested before, didn't it?

iShavgula, 2020-10-06:
Hello @DoNotGood,

I kind of found a solution: I replaced it with a LCD2004 display and I'm happy with it.

Sorry, but here is no hope(imho) for BTT002 + TFT35.

Anyways, I'm using PrusaSlicer + OctoPrint and rarely have to use the display. I may as well test the BTT wifi module in the future, which is a much cheaper alternative to OctroPrint.

Szarp78, 2020-10-15:
Hello @iShavgula , @DoNotGood 

To use Marlin mode, you must rotate the ribbon cables 180 degrees, but only on one side.

I have BTT002 + TFT43 and it works, but still, to change the firmware, you need to connect the old LCD2004 display.

DoNotGood, 2020-10-31:
Good morning @iShavgula,

I've tried your suggestion (turn around the cables about 180 °), and now the Marlin mode works, thx, but in Touch mode I have still no connection to the printer and I can't use the touch screen, it isn't reacting, it does not work, but when I stay with the finger on the screen, it switches to the menu, where I can choose Marlin or Touch mode, so it means the touch function of the display can't be totally dodgy!

iShavgula, 2020-10-31:
Good morning, 

Even with the LCD2004 I had to rotate the cables so that is board related, if I remember correctly that is mentioned in the manual pdf under hardware directory. But for me the only way to make the TFT work was without the reset pin (you need to connect only 4 cables instead of 5). Did you try that as well? 
[image]

DoNotGood, 2020-11-01:
Hi,

I've tried it (without RST-PIN), still it does not work, no connection no touch function, just Marlin mode.

gafillll, 2021-04-04:
switch exp1 exp2 180 and cut the nutch