bigtreetech/BTT002-V1.0 #10: sd crashing

dullenrrc, 2021-01-03:
i'm having some trouble with the board.

If i have an sd card inserted(lcd2004) while either resetting or powering on the board it crashes.
The printer functions ok when i insert the card after powering up.

this issue makes it quite difficult for me to update my firmware.

i have tried 3 different displays and a lot different sd-cards.

any ideas?

Thanks in advance!

Lino77, 2021-01-04:
I have it too. i solved the problem by using shorter cables to exp1 and exp2. these are max 20 cm long me the 60 cm long is not possible. I have a separate lcd2004 that I only swap for the bottle. there is already a fix or workout for it

stefanomn, 2021-03-28:
in my case an SD card prevents the printer from turning on.
and 99% of the time "with other SD card"
it is really impossible to do a firmware update even using multiple displays
I don't have the possibility to cut the flat cables to do other tests ... it would have been better to stay away from this product and take a skr 1.3 / 1.4 with the adapter cables ... now I would like a refund from the manufacturer ... hopefully