bigtreetech/BTT002-V1.0 #15: compatible with oficial prusa firmware?

sl1pkn07, 2024-06-21:
Hi

how luch difficult is make this board compatible with oficial prusa firmware?

greetings

3d-gussner, 2024-06-25:
@sl1pkn07 This is a 32-bit board and you can't use the 8-bit PrusaFirmware. I have seen Marlin 2.x configurations for this board.

sl1pkn07, 2024-06-25:
hi

im not talk about use the AVR arduino framework (8-bits) in this board. im talk about modify the prusa firmware for use with the STM arduino framework (32-bits)

greetings

3d-gussner, 2024-06-25:
@sl1pkn07 I don't think that is possible as the Prusa-Firmware is tailored for the Prusa MK3/S/+, MK2.5/S and diverted long long time ago from Marlin 1.x
If you want to use this board best chances are is to use Marlin 2.x and use that as vanilla Marlin 2.x can run on different hardware/electronics.

sl1pkn07, 2024-06-25:
ok. tnx

closed 2024-06-25 by sl1pkn07

3d-gussner, 2024-06-25:
Glad to help, even it is not the answer you wished for